keijiro/Akvfx #2: How to install

greengiant83, 2019-08-13:
This project looks like a fantastic jumping off point for Azure Kinect dev in Unity. Unfortunately, I am not sure how to install your files to get it working. 

Could you provide a couple pointers?

pixelpupinc, 2019-08-17:
Yeah,  I tried to run the project.  It is definitely accessing the camera, but it's not functioning as expected.  I tried changing a few settings and it did not seem to help.  Also I have to force quit the project after trying to pause playback in editor.  Were you using the official SDK or the Github beta?

keijiro, 2019-08-19:
Previously, this project used [Git support on Package Manager](https://forum.unity.com/threads/git-support-on-package-manager.573673/) that requires the Git client installed on the system.

I changed it to use the npm registry. Now it doesn't require any special setup. It should work if an Azure Kinect device is set up properly.

@greengiant83 Please start over from cloning this repository. This time it should work.

@pixelpupinc Please check if the device is properly set up by using Azure Kinect Viewer. If it works, this project should work too.

keijiro, 2019-08-19:
I'm closing this issue now. Please feel free to reopen it for further questions.

closed 2019-08-19 by keijiro

pixelpupinc, 2019-08-19:
Everything is working fine now.  I had the kinect viewer working OK, but re-pulling the project with the updates seemed to solve my issues as well.  This is so awesome, thank you for sharing.  Having the kinect settings already exposed as sliders/boxes is super helpful.

greengiant83, 2019-08-19:
This works great now! Thank you so much, man.