keijiro/Akvfx #5: 'ReadOnlySpan<>' could not be found

adrw2, 2019-09-20:
Hi
sorry, do i missing something? im getting this two errors. 

Assets\Akvfx-master\Assets\Akvfx\Internal\XYTable.cs(19,16): error CS0246: The type or namespace name 'ReadOnlySpan<>' could not be found (are you missing a using directive or an assembly reference?)

Assets\Akvfx-master\Assets\Akvfx\Internal\ThreadedDriver.cs(40,17): error CS0246: The type or namespace name 'ReadOnlyMemory<>' could not be found (are you missing a using directive or an assembly reference?)

thank you

closed 2019-09-20 by adrw2

amandalangley, 2019-10-10:
Any idea how you resolved the issue? I am having an issue with resolving dependencies.

adrw2, 2019-10-10:
importing his folder into an existing project doesnt work for me. I had to open directly his demo scenes.