keijiro/Akvfx #7: 502 Error

amandalangley, 2019-10-10:
Hey, thanks for the awesome work! When I open the project, I am having a problem with resolving dependencies.  I'm getting a 502 error. My Azure Kinect is working on my computer. I have the correct Unity version 2019.2.2f1.

Here is what happens when I open the project:
[image]

And this is what happens when I try and look at the Package Manager UI:
[image]

Any idea what is going on?

keijiro, 2019-10-10:
This project uses the [UnityNuGet](https://github.com/xoofx/UnityNuGet/blob/master/registry.json) registry to import some Nuget packages. It seems the registry site is temporarily unavailable.

sokobanni, 2019-10-22:
Looks like UnityNuGet doesn't work with 2019.2, but recent akvfx project is set for 2019.2.2f.
In my case, after I tried to run it with 2019.1.14f, it will works fine with 2019.2.

amandalangley, 2019-10-22:
> Looks like UnityNuGet doesn't work with 2019.2 (they said it only work with 2019.1), but recent akvfx project is set for 2019.2.2f.
> In my case, after I tried to run it with 2019.1.14f, it will works fine with 2019.2.

Hey, thanks, I've got it working now.

closed 2019-10-22 by amandalangley