keijiro/Akvfx #9: IL2CPP support

FernandoMonroy, 2019-11-27:
Does Azure support IL2CPP?

keijiro, 2019-11-28:
Azure? Did you mean Azure Kinect? (Azure is Microsoft's brand name for their cloud computing solutions)

Anyway, this project should support IL2CPP. Please report if you find any issue with IL2CPP.

keijiro, 2019-11-28:
I'm closing this issue now. Please feel free to reopen for further problems.

FernandoMonroy, 2019-11-28:
Sure Azure Kinect, sorry for the misleading.

I encounter problems with Intel real sense (using or not the Rsvfx repo) and old Kinect with unity compiling for IL2CPP. The compilations were successful but devices didn´t connect so I see no output of any kind (color or depth); Mono was the only way to run the devices. Is this the case of Azure Kinect? Does Azure Kinect connect properly using the IL2CPP scripting backend?

Don't have an Azure Kinect right now, but I´m considering buying one.

It´ll be super helpful if you could give me some feedback about it.

Thanks for your time.

keijiro, 2019-11-29:
Thank you for the additional information. I confirmed that the issue reproduces with IL2CPP. You can see the following error message when building with the development build option.

```
NotSupportedException: To marshal a managed method, please add an attribute named 'MonoPInvokeCallback' to the method definition. The method we're attempting to marshal is: Microsoft.Azure.Kinect.Sensor.Logger::OnDebugMessage
  at Microsoft.Azure.Kinect.Sensor.NativeMethods.k4a_set_debug_message_handler (Microsoft.Azure.Kinect.Sensor.NativeMethods+k4a_logging_message_cb_t message_cb, System.IntPtr message_cb_context, Microsoft.Azure.Kinect.Sensor.LogLevel min_level) [0x00000] in <00000000000000000000000000000000>:0 
  at Microsoft.Azure.Kinect.Sensor.Logger.Initialize () [0x00000] in <00000000000000000000000000000000>:0 
  at Microsoft.Azure.Kinect.Sensor.Logger.add_LogMessage (System.Action`1[T] value) [0x00000] in <00000000000000000000000000000000>:0 
  at Microsoft.Azure.Kinect.Sensor.AzureKinectOpenDeviceException.ThrowIfNotSuccess[T] (System.Func`1[TResult] function) [0x00000] in <00000000000000000000000000000000>:0 
  at Microsoft.Azure.Kinect.Sensor.Device.Open (System.Int32 index) [0x00000] in <00000000000000000000000000000000>:0 
  at Akvfx.ThreadedDriver.CaptureThread () [0x00000] in <00000000000000000000000000000000>:0 
  at UnityEngine.Application+LowMemoryCallback.Invoke () [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.ParameterizedThreadStart.Invoke (System.Object obj) [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <00000000000000000000000000000000>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x00000] in <00000000000000000000000000000000>:0 
  at UnityEngine.Application+LowMemoryCallback.Invoke () [0x00000] in <00000000000000000000000000000000>:0 
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
System.Xml.Schema.ValidationEventHandler:Invoke(Object, ValidationEventArgs)
```

I found the same issue in discord-rpc:

https://github.com/discordapp/discord-rpc/issues/186

As noted in this page, the issue may be easily solved by adding the `MonoPInvokeCallback` attribute to `Microsoft.Azure.Kinect.Sensor.Logger::OnDebugMessage`. However, it's in the Azure Kinect SDK code, so we have to send a pull request and get reviewed by Microsoft to apply the fix. That's a little bit, uh, troublesome.

At the moment, please understand that this project doesn't support IL2CPP. If you're motivated to solve the issue for some reason, please try the solution above.

FernandoMonroy, 2019-12-12:
Sorry for the late reply. Thank you very much for your feedback, I'll dig out into the subject