keijiro/Akvfx #10: All audio related components gone in Unity?

jelmerS2, 2019-12-27:
In this project, I can't get any audio to work. All audio related components are basically missing, for example there is no audio tab in the components menu, no audio listener can be added to the camera and in scripting also AudioSource is not recognized. What could be the reason for this? 
(Other than that this is an awesome repo)

keijiro, 2019-12-27:
They're missing because I removed them from the [package manifest file](https://github.com/keijiro/Akvfx/blob/master/Packages/manifest.json). You can recover them by copy-pasting from other projects like [this](https://github.com/Unity-Technologies/FPSSample/blob/master/Packages/manifest.json#L19-L48).

keijiro, 2019-12-27:
I'm closing this issue now. Please feel free to reopen it if it's still a problem.

closed 2019-12-27 by keijiro