keijiro/Akvfx #13: Is it real-time or rendering from a recorded data?

Lukikay, 2020-04-28:
Hi, this project is so awesome! I'm quite interested in it! 

I'm wondering if this is real-time with an Azure Kinect or it's just rendering from a recorded RGB/depth data?

And I'm also very interested in the VFX that from your projects Skinner/Smrvfx/DkvfxSketches that have more fancy particle effects. So are they using a similar framework and possible to have those VFX here?

keijiro, 2020-04-28:
Akvfx = realtime
Dkvfx = recorded (using Depthkit)

https://github.com/keijiro/Dkvfx

>  So are they using a similar framework and possible to have those VFX here?

Skinner - Legacy. Don't use it.
Smrvfx - VFX Graph
DkvfxSketches - VFX Graph

keijiro, 2020-04-28:
I'm closing this issue now. Please feel free to reopen it for further problems.

closed 2020-04-28 by keijiro