keijiro/Akvfx #15: XR Plugin Management Support

MrPantherM, 2020-06-22:
Using the Test scene, as soon as the XR Plugin Management package is installed, I receive error CS0246:

C:\Program Files\Unity\Hub\Editor\2019.4.1f1\Editor\Data\Resources\PackageManager\BuiltInPackages\com.unity.ugui\Runtime\UI\Core\InputField.cs(1798,17): error CS0246: The type or namespace name 'Event' could not be found (are you missing a using directive or an assembly reference?)

keijiro, 2020-06-22:
Please try closing and reopening the project. The error will disappear.

MrPantherM, 2020-06-22:
I can confirm the error disappeared. Thank you! 

Now trying to resolve issue of "black Screen" in headset. The headset renders properly on a test scene with out AkVFX. Im assuming its something to do with HDRP settings?

keijiro, 2020-06-22:
Maybe yes.

I don't have any experience on VR, so please don't expect that I can provide an effective answer/advice.

I'm closing this issue now as the original question was solved. Please open a new issue for another problem.

closed 2020-06-22 by keijiro