keijiro/Akvfx #16: ComputeBuffer SetData(ReadOnlySpan<byte>) causes error

felixfaire, 2020-06-23:
Hi Keijiro, firstly thanks for everything you give to the Unity community, secondly this may not be a direct issue, but more of a curiosity / issue i am facing:

I see that on line 115 of Runtime/DeviceController.cs you use:

`_colorBuffer.SetData(color.Span);`

to upload the image data to the compute buffer. I have been trying to use a similar technique to get kinect data onto the GPU (outside this package). I have all the Kinect libraries linking and devices opening fine but Unity (2019.3.14) will not allow this variation of ComputeBuffer.SetData causing an error:

cannot convert from 'System.ReadOnlySpan<byte>' to 'System.Array'

Could you please shed some light on how this code executes fine from within the package but causes an error from a script outside the package (in a new project).  Is this a mismatch in Unity / c# versions, or something to do with assembly compilation that I need to learn more about? Or if not how I can use a similar method to get the API 'Image' capture data into Unity Textures / Buffers without performing an expensive Span.ToArray() call which copies all the data.

Any help would be appreciated.

keijiro, 2020-06-23:
`SetData` is implemented in `Extensions.cs`, so you have to copy it into your project.

https://github.com/keijiro/Akvfx/blob/master/Packages/jp.keijiro.akvfx/Runtime/Internal/Extensions.cs#L44

felixfaire, 2020-06-23:
Aha! I dont know how i managed to miss that, makes a lot more sense. Thanks a lot.

closed 2020-06-23 by keijiro