keijiro/Akvfx #17: Crashes Unity Editor on recompile from script change.

felixfaire, 2020-06-29:
Hi Keijiro.

It seems that changing any user script while the player is running will cause the Unity editor to crash or hang until the kinect device is disconnected. I think this is something to do with the disposal and rejoining of the kinect driver thread, probably as it is still accessing a native plugin? Unfortunately my c# threading / GC knowledge is fairly limited so I cant immediately see what is wrong with the current code / method.

To reproduce crash:
- make a new empty c# script / monobehaviour
- run the app
- make a minor change to the script file and return to the editor.
- Editor crashes or hangs until the azure kinect is disconnected from the computer.

This happens regardless of the user settings for player behaviour on script changes.

unity 2019.3.14f1

felixfaire, 2020-06-29:
joining the thread and releasing resources on OnDisable() seems to avoid the crash but then extra null checking is required on Update().

keijiro, 2020-06-29:
I always avoid recompiling scripts while playing, so I couldn't find this problem.

At the moment, as a workaround, I'd recommend avoiding recompilation in the play mode.

felixfaire, 2020-06-29:
I try to avoid it too but sometimes forget. The crash still happens if you disable script recompilation while running (and select stop play first, or recompile after stop) which seems odd. 

My implementation with OnEnable and OnDisable seems to get around the issue for now but I wish there were more places to read about what unity does when a script file changes or how to properly / safely integrate external API's like the Azure Kinect.Sensor

Thanks