keijiro/Akvfx #18: Need help with running

RollySeth, 2020-07-04:
Hi,

Thanks for sharing. 
I am having issues with running it. Can you share some more information on how to use it?

This is what I am currently doing. 
I downloaded the git folder and opened in Unity 2020. Got three errors- 
1.  InputField.cs(1628,40): error CS0246: The type or namespace name 'Event' could not be found (are you missing a using directive or an assembly reference?),
2. error CS0246: The type or namespace name 'Event' could not be found (are you missing a using directive or an assembly reference?),
3. InputField.cs(1628,40): error CS0246: The type or namespace name 'Event' could not be found (are you missing a using directive or an assembly reference?)

Also, none of the scripts are loaded. 

I tried 'Reimporting All' , changing Event to UnityEvent / UnityEngine.Event  in InputField.cs (that removed above errors but started showing several other errors in InputField.cs eg. InputField.cs(1230,24): error CS0103: The name 'GUIUtility' does not exist in the current context
), removed Packages (with manifest file) folder for unity to regenerate it (it removed above errors but nothing happens when hit run command in Unity). 

I am comparatively new to Unity and thus, any insights on how to try out your code will be highly grateful.

keijiro, 2020-07-05:
Please use Unity 2019.3 or 2019.4 at the moment. There is some small compatibility issues. I'll fix it when 2020 is officially released.

keijiro, 2020-07-05:
I'm closing this issue now. Please feel free to reopen it if it still reproduces.

closed 2020-07-05 by keijiro