keijiro/Akvfx #19: Error running test

CaptainMetropolis, 2020-07-27:
Trying to get this setup. I'm running into the same issue as others have reported:

"error CS0246: The type or namespace name 'Event' could not be found (are you missing a using directive or an assembly reference?)"

However, none the other fixes seem to work. I'm using Unity 2019.4.5f1 and tried restarting Unity after installing the XR Plugin Manager. Tried cloning the project again - no luck. 

Any suggestion on what to try?

keijiro, 2020-07-27:
Could you try Unity 2019.4.3? There is a small compatibility issue between 2019.4.x. There are some workarounds, and the easiest way to do it is downgrading the Unity version.

keijiro, 2021-10-07:
I'm closing this issue due to inactivity. Please feel free to reopen it for further problems.