keijiro/Akvfx #20: Reading only color frame

xubi1993, 2020-09-07:
Hello,
         Great asset, works very well but can you guide me how can I get the 2D camera color image inside the vfx graph. Here it computes the depth frame and depth color frame. I just want a normal color frame with no depth. How can I compute that. Is is even possible? Sorry I am not very good with shaders. Any help will be much appreciated.

Regards,
Zubair

keijiro, 2020-09-07:
You can simply sample the color map using the Sample Texture2D operator.

closed 2020-09-07 by keijiro

xubi1993, 2020-09-08:
I tried it but it does not appear like it does in 3D. It does not create a plane of 2D particles with the image from camera mapped on top of those particles. For example like this: https://www.youtube.com/watch?v=l5mZgdNeZD4

xubi1993, 2020-09-08:
Thanks I got it, I just need to set the UV's right!