keijiro/Akvfx #21: Changing camera field of view

DangerDay, 2020-10-12:
Moving this question over from AKVJ.

Looking for a way to set the Azure Kinect to 120 degree FOV, found this bit here in ThreadedDriver.cs:

new DeviceConfiguration
{ ColorFormat = ImageFormat.ColorBGRA32,
ColorResolution = ColorResolution.R1536p, // 2048 x 1536 (4:3)
DepthMode = DepthMode.NFOV_Unbinned, // 640x576
SynchronizedImagesOnly = true });

Along with these settings here:

public static int ImageWidth => 640;
public static int ImageHeight => 576;

When I try to match what should be the WFOV_2x2Binned or WFOV_Unbinned settings with the ImageWidth and ImageHeight I either get a device error or it will start up but I get some pretty bad lines across the VFX graph effects.

My question is are there more settings that I haven't found yet, or is there another way to set the field of view?

keijiro, 2020-10-13:
I think this line must be modified too:

https://github.com/keijiro/Akvfx/blob/master/Packages/jp.keijiro.akvfx/Resources/Akvfx.compute#L26

keijiro, 2020-10-14:
According to the hardware specification, the WFOV Unbinned mode doesn't support 30 FPS.

https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification

So you may have to change the FPS to 15.

https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/class_microsoft_1_1_azure_1_1_kinect_1_1_sensor_1_1_device_configuration_a68756d38f2bc49df2dc9316fa88e8115.html#a68756d38f2bc49df2dc9316fa88e8115