keijiro/Akvfx #23: When I try to open the project I got an error window.

o986o, 2020-11-05:
When I try to open the project I got an error window. 
I open the project in Unity 2019. 3. 14f1.

when I open the project at first I got this error window. 
[image]

when I click continuew button, I can see the below errors on the console. 
[image]

I wonder how can I solve these problems.

keijiro, 2020-11-06:
This error was caused by the UnityNuGet registry. The issue has been already fixed. Clear the Package Manager cache directory and retry opening the project.

closed 2020-11-06 by keijiro

o986o, 2020-11-06:
> This error was caused by the UnityNuGet registry. The issue has been already fixed. Clear the Package Manager cache directory and retry opening the project.

Thanks for your fast reply. 
But I still have an error message window. 

The symptom is this. 

[image]

I delete my package manager cache and also I deleted manifest.json.
[image]
[image]

[image]

keijiro, 2020-11-06:
You shouldn't delete the files in the project directory, like manifest.json.

You only have to clear the cache under `user/AppData`

o986o, 2020-11-06:
> You shouldn't delete the files in the project directory, like manifest.json.
> 
> You only have to clear the cache under `user/AppData`

Thanks a lot, Keijiro san. I success to open the project. 

If someone has the same problem as me, just delete the package cache as under image.
The directory is this.
> C:\Users\user\AppData\Local\Unity\cache
I deleted everything in the Cache folder.
 
I hope this helps you. 

[image]