keijiro/Akvfx #28: Issue importing to 3rd party project

Changliu52, 2020-12-29:
Hi @keijiro 

We are trying to import the unity package exported from this project. and at the same time added
```
{
  "scopedRegistries": [
    {
      "name": "Unity NuGet",
      "url": "https://unitynuget-registry.azurewebsites.net",
      "scopes": [
        "org.nuget"
      ]
    },
    {
      "name": "Keijiro",
      "url": "https://registry.npmjs.com",
      "scopes": [
        "jp.keijiro"
      ]
    }
  ],
  "dependencies": {
    "jp.keijiro.kino.post-processing": "2.1.15",
    "jp.keijiro.klak.motion": "1.0.1",
    "jp.keijiro.metamesh": "0.0.6",
    "com.unity.modules.animation": "1.0.0",
    "com.unity.modules.imageconversion": "1.0.0",
    "com.unity.modules.physics": "1.0.0",
    "com.unity.modules.uielements": "1.0.0"
  }
}
```
We are testing with an empty project on unity 2019.4.8f1
all game objects in the Test.scene looks fine in the inspector. But the point cloud was not rendered. No error in the console.

The same setting works fine in the direct clone of this project.

Please help.

Chang

keijiro, 2020-12-29:
Could you share a minimal project that reproduces the issue?

ly19264, 2020-12-29:
Hi @keijiro ,

A minimal project could be found at https://github.com/ly19264/KinectIntegration for the reproduction of the issue

ly19264, 2021-01-06:
Hi @keijiro ,

We were able to integrate with 2019.3.9f1. It turns out the project has dependency on .NetStandard2.0 , would it be possible to support .Net 4.x ?

keijiro, 2021-01-06:
@ly19264 No. Context: https://github.com/keijiro/KlakNDI/issues/72