keijiro/Akvfx #29: enabling recorder

LensNinja-Aleksa, 2021-03-29:
Ola! 
I can't seem to be able to install the recorder whilst in this project. I am trying to install the 2.5.5 version, which I could install on a new project in Unity 2019.4.8f1. 
When I am in this AKVFX project and try to install recorder, it gives me this red message "Library\PackageCache\com.unity.timeline@1.2.6\Runtime\Audio\AudioTrack.cs(29,40): error CS0246: The type or namespace name 'AudioClip' could not be found (are you missing a using directive or an assembly reference?)"

Does anyone know how to resolve this issue and enable the recorder for this, please?

keijiro, 2021-03-30:
You have to enable the built-in Audio package on the Package Manager UI.

keijiro, 2021-03-30:
I'm closing this issue now. Please feel free to reopen it for further problems.

closed 2021-03-30 by keijiro