keijiro/Akvfx #33: Running Multiple Sources Simultaneously

ianmcdermott, 2022-07-18:
Hi Keijiro,

Great plugin! This may not be an issue, I may just be doing something wrong, but is it possible to use multiple sources simultaneously for multiple vfx renders? When I try running two mkv recordings, it keeps swapping the textures between the sources. Might just be a limitation so figured I'd ask.

Thanks!
Ian

keijiro, 2022-07-19:
I haven't seen this project for over two years, so I can't remember it exactly, but I think the current implementation only supports a single source. You may have to modify the device controller class to support multiple sources.

ianmcdermott, 2022-08-03:
Thanks so much Keijiro! I ended up not needing to do it programmatically. I didn't realize I had to designate separate depth and color textures to each VfxPointCloud/Kinect. Again, really great plugin. Thanks for all your work!

closed 2022-08-03 by ianmcdermott