keijiro/Akvfx #35: Getting point cloud soup when streaming textures

thePostFuturist, 2023-02-19:
Hey @keijiro ,

I'm trying to use your setup with [FMETP ](https://assetstore.unity.com/packages/tools/video/fmetp-stream-3-0-221362)to stream the two point cloud textures over the network, and getting some distortions when trying to process the textures.  A regular VFX Property Bind drawing from the Kinect produces results per spec, but when pulling it over the network produces oddities as illustrated in the screenshot. 

[image]

The script I used is a mod of your VFXPointCloudBinder, just drawing Texture components from a Decoder instead of the Kinect device. LMK if you need any more info. Nothing wrong with your current project, but figured you might shine some light on what's going on here. Thanks for the package.

keijiro, 2023-02-19:
Does FMETP support `ARGBFloat` texture? I have never tried it, but I don't think it supports non-standard color formats.

thePostFuturist, 2023-02-19:
According to the dev:

Unfortunately, only 8bit(0-255) is supported.
Thus, you may have data loss for 32bit float format.

Is there a way to accommodate for 8 bit?

keijiro, 2023-02-19:
> Is there a way to accommodate for 8 bit?

The short answer is "no". As you may know, 25% fixed rate compression (32-bit data into 8-bit data) is quite challenging. That's out of the scope of this project.

keijiro, 2023-02-19:
I'm closing this issue as answered. Please reopen it for further problems.

closed 2023-02-19 by keijiro