keijiro/Akvfx #38: Kinect V2 help

burhanahmad02, 2025-03-28:
@keijiro Thankyou dear for clearing that the project will not be compatible with kinectv2 as its different from Azure Kinect, I would be so thankful if you could guide or hint me to the path where I can achieve the same result as in the video with Kinectv2

https://github.com/user-attachments/assets/4b58e438-4f45-4ff4-b3c8-860eab7634a3

keijiro, 2025-03-28:
Sorry, but please understand that this issue tracker is dedicated to this specific project (Akvfx) and not a general consulting forum for other projects. Additionally, I have not worked with Kinect v2, so I am unable to provide guidance on achieving similar results with it.

closed 2025-03-28 by keijiro

burhanahmad02, 2025-03-28:
Thankyou